mathertel/Radio #3: Volume not working

sn4k3, 2015-07-16:
Hello,

I find this library great! Still i can't get volume to work with your demo serial or fixed...
Im using a TEA5767 but not the simple one, im using this one: http://i.ebayimg.com/00/s/NzY4WDEwMjQ=/z/X18AAOSwPhdVDcz0/$_57.JPG

Does volume should work this this module?

mathertel, 2015-07-16:
Hi Tiago,

 

The TEA5767 doesn’t have any volume control inside and the audio output is always full speed or mute.

I guess that the other chip on that Breakout is a TDA1308. This amplifier cannot be volume controlled.

So: NO.

 

Regards, Matthias

sn4k3, 2015-07-18:
Hi,

So there are any radio module with volume control? What do you recommend?
Maybe i can modify the TEA5767 and add a Digital Potentiometer to control de output?

Thanks

nikolisan, 2016-04-30:
Hi Thiago, I'm using a SI4703 breakout board I've found cheap on ebay. It has volume control and RDS. If you ensist on using the TEA5767 you can make your own amplifier and adjust the volume from there with a digital or analog pot. Search on google and you'll find many instructables.

sn4k3, 2016-04-30:
Hello, i solve it with a digital pot some month ago :)
Thanks

nikolisan, 2016-04-30:
Glad to hear that! May I ask how? You used TEA5767?

sn4k3, 2016-04-30:
Sure, check this schemenatic: https://github.com/sn4k3/LedClock/blob/master/Eagle/ledclock_schemantic.png
I used DS1803 digital 100K Pot

sn4k3, 2016-04-30:
And: https://github.com/sn4k3/LedClock/blob/master/Arduino/RadioFM.cpp

closed 2019-06-28 by mathertel