mathertel/Radio #4: Doesn't works

Raouloq, 2015-11-15:
Hi
First, thx for your work
I've this TEA5767 from here:
http://fr.aliexpress.com/item/TEA5767-FM-Stereo-Radio-Module-for-Arduino-76-108MHZ-With-Free-Cable-Antenna/32285486392.html?adminSeq=201112132&isOrigTitle=true&shopNumber=110055
It's connected to the arduino uno, and i upload the sketch with the arduino ide 1.6.3 and it doesn't works. Everytime there is just noise and for TestTEA5767 i've this:

Radio...
setFrequency(9540)
*11672
Write Fail:2
setVolume()
setMono()
Write Fail:2
Station:362.58 MHz
Radio: --- -----  MONO   RSSI: 0
Audio: ---- ---- --------
Station:362.58 MHz
Radio: --- -----  MONO   RSSI: 0
Audio: ---- ---- --------
Station:362.58 MHz
Radio: --- -----  MONO   RSSI: 0
Audio: ---- ---- --------
Station:362.58 MHz
Radio: --- -----  MONO   RSSI: 0
Audio: ---- ---- --------
Station:362.58 MHz
Radio: --- -----  MONO   RSSI: 0
Audio: ---- ---- --------

I tried with others library and no one works, there is just 1 time it switched on a radio and it was very clear, i tried to change again but no succes, i think it was just chance.
I noticed in the noise we can hear the keyboard and sometimes the arduino when it send data to the computer.
Maybe it come from the arduino version ?

mathertel, 2015-11-15:
Hi ,

 

it seems to be a problem in the I2C connections to the chip or the chip itself because the frequency you set is not reported back.

You should use the TestTEA5767 example first and check the wiring as mentioned in the sketch file:

Arduino port | TEA5767 signal

------------ | ---------------

        3.3V | VCC

         GND | GND

          A5 | SCLK

          A4 | SDIO

          D2 | RST

I hope for you that the chip is not broken…

Regards, Matthias

 

http://fr.aliexpress.com/item/TEA5767-FM-Stereo-Radio-Module-for-Arduino-76-108MHZ-With-Free-Cable-Antenna/32285486392.html?adminSeq=201112132 <http://fr.aliexpress.com/item/TEA5767-FM-Stereo-Radio-Module-for-Arduino-76-108MHZ-With-Free-Cable-Antenna/32285486392.html?adminSeq=201112132&isOrigTitle=true&shopNumber=110055> &isOrigTitle=true&shopNumber=110055

Raouloq, 2015-11-16:
Thank you for the information, i think you have right about the L2C. I tried with 3.3v but normaly i use 5v(I saw we need to use 5v when it's with the board), but there is no rst pin. I tried with an another arduino(mega) and it change nothing. The radio worked when i touched the antenna(and surly a part of the board) so maybe there is some problems on the soldering, i need to see that.

Raouloq, 2015-11-16:
The soldering is normal, so it's maybe on the chip :/

rohanxtreme, 2016-12-21:
Hi Raouloq,
Did you get to the bottom if the issue you were facing ?
I'm creating a similar project at a much simpler scale and with RPi3 and TEA5767 .

I'm also just getting static noise with some breaks when i type.

Was it the code ? Or was it the physical connections ?

Raouloq, 2016-12-21:
Hi,

Nope :/ I think it was the board and not the code, i didn't bought an another board for be sure because it was for a school project with a deadline and it was not possible to wait 2 anothers weeks. But i still have the board, i will try again soon for be sure

closed 2019-06-19 by mathertel

adrieng1977, 2023-07-02:
had the same issue
read this
https://www.arduino.cc/reference/en/language/functions/communication/wire/
I have the MEGA 250

Mega2560 | 20 (SDA), 21 (SCL)

When switching to the right ports it started working