mathertel/Radio #5: RDA5807 have problem with seekUp & seekDown functions.

VinokurovSE, 2016-01-07:
Hello, i have troubles.
The bottom line: To use the search function all that I use to adequately respond to my requests: turn the sound off, set the volume, frequency, no complaints at work.
BUT. All this works properly before use functions and seekUP seekDovn.
The function is activated, a new station is playing it, but if you use any other functions of the radio again begins looking for a new frequency ... change the volume - and changing the volume and frequency ...
It's the same with the example included with the library.
To solve the problem can not give access to the PC and connected to the Arduino module. Sound will bring the microphone. All for debugging, so to speak.
And further. In our forums (I'm from Russia), I found the information that this module you can switch the antenna input. I can tell you more, if you're interested.
Thank you in advance!

RicoElectrico, 2017-02-09:
@VinokurovSE I also face the same problem. Did you fix it or what? I could not understand what you mean.

planer-pro, 2017-02-24:
same trouble!

planer-pro, 2017-02-24:
It seems, where is not complete working library for this module. Module will work fine if use it direct wire. Many examples in internet.

wmarkow, 2017-11-04:
>The bottom line: To use the search function all that I use to adequately respond to my requests: turn the sound off, set the volume, frequency, no complaints at work.
BUT. All this works properly before use functions and seekUP seekDovn.
The function is activated, a new station is playing it, but if you use any other functions of the radio again begins looking for a new frequency ... change the volume - and changing the volume and frequency ...
It's the same with the example included with the library.

I have the same problem with **RDA5807M** chip. After power up the main functions (like setVolume, setMute, setBassBoost, and so on) work fine until **seekUp** or **seekDown** is executed. After that **setMute** and **setBassBoost** trigger additional **seekUp** or **seekDown**. This is what I have noticed.

I think it is related to the cached register values in the library: **registers[RADIO_REG_CTRL]**. The library remembers that a call to **seekUp** or **seekDown** has been executed and sets the **SEEK** bit in the cached **RADIO_REG_CTRL** register. This bit is not cleared later and when calling **setMute** (for example) the chip is being instructed to seek up (or down) again. But, the **setMono** method is immune to this issue because it clears the **SEEK** bit before the mute will be send.

closed 2020-04-23 by mathertel