mathertel/Radio #13: suggestion: Decoding the date information with MJD!

ClashC, 2016-12-26:
Hello,
I use for my project the decoding of the date information about MJD.
This information i have from here: http://pic-projekte.de/phpBB3/viewtopic.php?f=14&p=7475&#p7458
Maybe you like it too? :)

Example:
```
void DisplayTime(uint8_t hour, uint8_t minute, uint32_t mjd) {
  //SetRTCTime(hour, minute);
  setRTCdatefromMJD(mjd);
} // DisplayTime()
...
void setRTCdatefromMJD(uint32_t mjd)
{
  int k;

  int _year = (int) ((mjd - 15078.2) / 365.25);
  int _month = (int) ((mjd - 14956.1 - (int)(_year * 365.25)) / 30.6001);
  int _date = (int) (mjd - 14956 - (int)(_year * 365.25) - (int)(_month * 30.6001));

  k = (_month == 14 || _month == 15) ? 1 : 0;
  _year = _year + k;
  _month = _month - 1 - k * 12;
  //_month--;

  long WD=(long)((mjd+2) % 7)+1; //modulo 7 | Day of the Week
... Code ...
}
```

Patch:
```
--- ./src/RDSParser.cpp 2016-12-22 21:55:19.647046114 +0100
+++ ./src-neu/RDSParser.cpp     2016-12-17 13:06:56.000000000 +0100
@@ -58,6 +58,7 @@

   uint16_t mins; ///< RDS time in minutes
   uint8_t off;   ///< RDS time offset and sign
+  uint32_t MJD;

   // Serial.print('('); Serial.print(block1, HEX); Serial.print(' '); Serial.print(block2, HEX); Serial.print(' '); Serial.print(block3, HEX); Serial.print(' '); Serial.println(block4, HEX);

@@ -150,7 +151,14 @@
     off = (block4)& 0x3F; // 6 bits
     mins = (block4 >> 6) & 0x3F; // 6 bits
     mins += 60 * (((block3 & 0x0001) << 4) | ((block4 >> 12) & 0x0F));
+
+       // und jetzt modifiziertes julianisches Datum berechnen
+       MJD=(block3 >>1);                                       //rechts schieben, da Bit0 nicht relevant ist
+       MJD=(MJD|0x4000) | ((block2 & 0x1)<<14);                //Bit 0 von Block B wird Bit15 in MJD
+       MJD=(MJD|0x8000) | ((block2 & 0x2)<<15);                //Bit 1 con Block B wird Bit16 in MJD
+
     // adjust offset
     if (off & 0x20) {
       mins -= 30 * (off & 0x1F);
@@ -160,7 +168,7 @@

     if ((_sendTime) && (mins != _lastRDSMinutes)) {
       _lastRDSMinutes = mins;
-      _sendTime(mins / 60, mins % 60);
+      _sendTime(mins / 60, mins % 60, MJD);
     } // if
     break;

diff -Nur ./src/RDSParser.h ./src-neu/RDSParser.h
--- ./src/RDSParser.h   2016-12-22 21:55:19.647046114 +0100
+++ ./src-neu/RDSParser.h       2016-12-17 12:52:22.000000000 +0100
@@ -28,7 +28,7 @@
 extern "C" {
   typedef void(*receiveServicenNameFunction)(char *name);
   typedef void(*receiveTextFunction)(char *name);
-  typedef void(*receiveTimeFunction)(uint8_t hour, uint8_t minute);
+  typedef void(*receiveTimeFunction)(uint8_t hour, uint8_t minute, uint32_t MJD);
 }

--- ./src/radio.h       2016-12-22 21:55:19.653046114 +0100
+++ ./src-neu/radio.h   2016-12-04 13:49:26.000000000 +0100
@@ -54,6 +54,7 @@

 #include <Arduino.h>
 #include <Wire.h>
+#include <stdarg.h>

```

getSurreal, 2020-03-25:
When I try to use this, I'm not getting back an accurate year, although month and day are good.  

`mjd = 58933`

should result in 3/25/2020 but the year is showing as 120.  Is there additional conversion that should be applied to it?

getSurreal, 2020-04-07:
After further research I discovered that you add 1900 to the mjd year value.

getSurreal, 2020-04-07:
This PR implements this suggestion.  https://github.com/mathertel/Radio/pull/38