mathertel/Radio #14: RDA5807M + Arduino Pro Micro + SerialRadio Example = Random Freezing

MatPeff, 2017-05-31:
Hello,
At the beginning thank you for this work , It's very interesting project and comprehensive solution to easy  implement of SDR receivers.
I have a problem like this : 
I connected typical RDA5807 module to Arduino Micro and have uploaded example from your library - SerialRadio. Everything looked good , i could give serial commands to the radio( like frequency change (fxxxxx) , seekUp , seekDown , volume +/- ) , i could receive RDS messages but after few minutes I lost communication with radio via serial. 
In other case i have connected a LCD with I2C converter and encoder to manually control of radio(change frq , vol etc...) . Frequency , and RDS was displayed properly , I could control radio by encoder but after few minutes everything froze again and I couldn't control radio by encoder. RDS refreshing was frozing too. I2C Addresses of LCD and RDA are different , I think that isn't a problem. In my custom implementation I turned off a serial communication. 
I have tried to comment parts of code with control encoder but It still isn't It and It did not help.

Below is my implementation of your code with LCD control : 

#include <EEPROM.h>
#include <newchip.h>
#include <radio.h>
#include <RDA5807M.h>
#include <RDSParser.h>
#include <Wire.h>
#include <LiquidCrystal_I2C.h>
#define outputA 6
#define outputB 7
#define zeroButton 8
int counter = 0; 
int aState;
int aLastState;  
int n = LOW; 
unsigned long swTime = 0;
bool turnRight = false;
bool turnLeft = false;
int mode = 0;
int volumeEnc = 0;
int volumeSet;
RADIO_FREQ fqToSave;

byte Vol1[8] = {
  B00000,
  B00000,
  B00000,
  B00000,
  B00000,
  B00000,
  B00000,
  B11111,
};

byte Vol2[8] = {
  B00000,
  B00000,
  B00000,
  B00000,
  B00000,
  B00000,
  B11111,
  B11111,
};

byte Vol3[8] = {
  B00000,
  B00000,
  B00000,
  B00000,
  B00000,
  B11111,
  B11111,
  B11111,
};

byte Vol4[8] = {
  B00000,
  B00000,
  B00000,
  B00000,
  B11111,
[... 371 lines cut ...]
        if (i_sidx > 0) {
          i_sidx--;
          radio.setFrequency(preset[i_sidx]);
          int fq = radio.getFrequency();
          DisplayFrequency(fq);
        }
      }
     }     
   }
   aLastState = n; 
   radio.checkRDS();
  if (now > nextFreqTime) {
    f = radio.getFrequency();
    if (f != lastf) {
      // print current tuned frequency
      DisplayFrequency(f);
      lastf = f;
    } // if
    nextFreqTime = now + 400;
  } // if  
}

Could you help me with this problem ? I would be very grateful for your help :) 
Best regards.
Matthew.

wmarkow, 2018-03-29:
@MatPeff, do you still have this freezing problem? Maybe you suffer to this arduino/Arduino#1476 issue in **Wire** library? Take a look at my wmarkow/arduino-hifi#3 where I describe my problems with a blocked I2C; maybe it will help you. At least you can use [ArduinoCrashMonitor](https://github.com/Megunolink/ArduinoCrashMonitor) to find the place in the code which is responsible for freezing.

mathertel, 2023-02-03:
closed as there was no activity.

closed 2023-02-03 by mathertel