mathertel/Radio #25: New Arduino IDE doesn't compile the Examples

roboticboyer, 2019-06-15:
I'm trying to compile the examples of the Radio library without any success
There are a lot of errors in the library probably because the Arduino IDE has been updated after revision 1.6 with changes not compatible with this library
E.g. Arduino.h is not yet required, RADIO.h and radio.h are different, etc.
@mathertel please could you update the library to be compatible with the new Arduino IDEs
Thank you
Gio

Phoenix1747, 2019-06-15:
Would do a pull request for the old library includes in the examples. But it looks like there is more stuff broken in some examples, like the analog pin numbers. Changing this rather randomly would maybe change what @mathertel has in mind about them.

roboticboyer, 2019-06-17:
My request is that @mathertel will kindly update the library. I haven't the a so big knowledge to do the changes
Are you @Phoenix1747 able to do the changes and propose the pull request or a fork?
Thank you
Gio

Phoenix1747, 2019-06-17:
I could do a pull request for most of the lib examples in like two weeks. No time right now, though.

roboticboyer, 2019-06-18:
I made further investigation
The compilation errors are related as minimum to:
- it's required the library Arduino.h not arduino.h
- it's required the library radio.h not RADIO.h

The above errors are shared everywhere in the .cpp and .h files in the /src directory and in the examples .ino file

Probably there is also the #22 issue

Unfortunately I'm not yet able to do a pull request or fork the library with my fixes

mathertel, 2019-06-19:
should be fixed with release 1.1.0.

closed 2019-06-19 by mathertel

roboticboyer, 2019-06-20:
It's required changing version in library.properties so the Arduino Library Manager bot will update the libraries repository

[link](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list)

Thank you
Gio