mathertel/Radio #27: No need to wait after Wire.requestFrom() and it should not be followed by Wire.endTransmission().

Koepel, 2019-06-28:
In the file "Radio/src/SI4703.cpp" the waiting after a `Wire.requestFrom()` may be removed. It is this line: `while(Wire.available() < 32) ;`

In the file "Radio/src/RDA5807M.cpp" the `Wire.requestFrom()` is followed by a `Wire.endTransmission()` in four places. Those `Wire.endTransmission()` may be removed.

Explanation: [Common-mistakes](https://github.com/Koepel/How-to-use-the-Arduino-Wire-library/wiki/Common-mistakes), number 1 and 2.

mathertel, 2019-06-28:
added changes to the code. Waiting for test results before creating new version.

closed 2019-06-28 by mathertel

wmarkow, 2019-06-28:
@mathertel, if you do not mind I would ask here a question to @Koepel.

@Koepel, it looks like you have some knowledge of **Wire** library. There is a known issue https://github.com/arduino/ArduinoCore-avr/issues/42 inside of **Wire** where the execution of the code may freeze in those "infinite" **while** loops. 
In my local branch https://github.com/wmarkow/ArduinoCore-avr/tree/issue_%2342 I have made some proposition of the fix (based on the work of some other volunteers):
https://github.com/wmarkow/ArduinoCore-avr/commit/8b37c0685ffdaf473dbdf8e7661fd6c991aad196
There is also a pull request https://github.com/wmarkow/ArduinoCore-avr/pull/1 with some modification of my changes.

Will you have some time to take a look at this and say if this makes sense? At least with those changes, the freeze issue is not enymore reproducible in my test project.