mathertel/Radio #29: How the web pages are generated?

CGarces, 2019-10-21:
I see the `params.json` at https://github.com/mathertel/Radio/commit/b34d6f327485ccf62982aa06bc084fc1870d5365
But I don't get how is used I can't see any script.

mathertel, 2019-10-21:
This file is part of the gh-pages , it gets generated by the doxygen tool (www.doxygen.org<http://www.doxygen.org>).
The `html` folder contains the sourcecode documentation that can be seen at: http://mathertel.github.io/Radio/html<file:///C:/Users/Matthias/Projects/Arduino/Sketches/Libraries/Radio/html/html/index.html>

I am not sure how this file is used. I guess by the doxygen tool.

I see the params.json at b34d6f3<https://nam04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmathertel%2FRadio%2Fcommit%2Fb34d6f327485ccf62982aa06bc084fc1870d5365&data=02%7C01%7C%7Cfac950b045b94b44782a08d75651cee3%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637072781668733172&sdata=VpkI9eZ5hDllVsBhqc%2F%2B1dfaobbJd5ZWa%2FxQv3v3MJ4%3D&reserved=0>

CGarces, 2019-10-21:
All the doxygen stuff is generate at html folder. 
I have no idea about the files outside that folder like `params.json` or `index.html` i can't find any reference on master.

I'll send you 3 different PR to 

- Build the web page using Jekyll themes
- Sync README.md from master and index.html from gh-pages
- Rebuild doxygen documentation and deploy to gh-pages after any commit on master.

CGarces, 2019-10-21:
Ok, that files come from the old "GitHub’s page generator "
http://webcache.googleusercontent.com/search?q=cache:QIeygz-FtYQJ:help.github.com/en/enterprise/2.16/user/articles/creating-pages-with-the-automatic-generator+&cd=1&hl=es&ct=clnk&gl=es

And was replaced some years ago by Jekyll

closed 2019-10-21 by CGarces