mathertel/Radio #33: Si4721 TxRx

NPoole, 2019-11-22:
How would you feel about a pull request that adds transmit support via Si4721?

mathertel, 2019-11-23:
That sounds great. I like to support you on this.
Can you send me more details on your plan ?

NPoole, 2019-11-30:
As for adding the radio to the library as a receiver, it should be pretty straight-forward. I can basically clone the SI4705 files to make SI4721.cpp and SI4721.h 

The transmit part is what I'm still trying to decide on. I have working transmit code, but I don't know how to integrate it into the library. Maybe you have an idea? 

In order to put the device into Tx mode, you need to issue the power-down command and then bring it back up and initialize in transmit mode. It's possible a user would want to switch the same radio module back and forth, which works fine, but could be confusing considering that (as of now) this is the only transmit-capable module in the library. 

Here's my initial plan: 

-Add a canTransmit() or transmitCapable() function to radio.cpp that returns true only if a transmit mode function has been defined. This enables example code to check for transmit capability before trying to reset the device.

-Add a switchMode(tx/rx) function to SI4721.cpp that resets the device and initializes it in the selected mode. Also, store the current mode in memory to reference later.

-Modify setFrequency() to set the transmit frequency if in transmit mode

-Add transmit-specific functions to SI4721.cpp (set RDS text out, set RDS program name, etc)

Can you think of any reason this would be an undesirable approach?

NPoole, 2019-12-10:
I believe my fork is working. I have to do some documentation and make some stylistic changes but I should be making a pull request soon.

henkiejan1, 2020-01-01:
Hmm doesn´t like he accept the pull request? Should yhe Si4721 work with a esp8266 or esp32? I can´t get it work with the Si4703...?

mathertel, 2020-01-04:
Please be patience. I will check for compatibility in mid-January.

mathertel, 2020-09-17:
@NPoole: I am missing some code in the Transmit examples. You do not call the setModeXXX in here. Maybe you have another version of the example locally.

In the meantime I finished the receiving mode in branch: https://github.com/mathertel/Radio/tree/feature_Si4721

NPoole, 2020-09-20:
Damn, I'm sorry about that. It turns out my library directory on my laptop was not pointing at the github repo folder so everything got out of sync. I've just made a commit to my fork that adds proper example code.

closed 2020-09-20 by mathertel

synth, 2021-01-03:
@NPoole I am interested in such a breakout board! Is this going to be released by Sparkfun? Is this going to be a replacement for the NS73m board Sparkfun discontinued?