mathertel/Radio #46: SI4703

Andr1961, 2020-11-07:
Hi,
I ran an example on the ESP32 ScanRadio.ino
but only noise is heard, when scanning, it catches nothing.
my settings
#ifdef ESP32
int resetPin =19; // rst
int SDIO = 21; //sda
int SCLK = 22;  //scl
int STC = 18;   //gpio2
#endif //ESP32
SI4703 radio(resetPin,SDIO);

in setup
#ifdef ESP32
Wire.begin(SDIO, SCLK);
#endif
the library from SparkFun works well.
Advise what can be done, the library is good because universal, thank you.

Beppi4U, 2020-11-28:
Scan Radio Sketch workes fine!

Andr1961, 2020-11-28:
ScanRadio.ino

Radio...>init()
Write Fail:2
Write Fail:2
>setBand(1)
>setFrequency(10370)
Write Fail:2
>_waitEnd()
>Seek limit hit
Write Fail:2

I'm using the library now
https://github.com/dmitryelj/Si4703-Arduino-Sketch-For-RDS-Spy
started immediately without problems,
RDA5807 on esp32 your library is working fine.

mathertel, 2023-02-07:
ESP32, Scanradio and starting Wire was fixed in 3.0.1
Wire->begin must not be called before reset sequence. 

Thanks for pointing me to RDS-Spy

closed 2023-02-07 by mathertel