mathertel/Radio #47: Just noise LCDRadio SI7403

Beppi4U, 2020-11-28:
Hi there!
Build a LCDRadio with SI4703 a16x2LCD via I2C and a rotary enconder.
Platform is a Arduino nano clone 328p.
I used the LCD Radio Sketch as shown down.
System starts up. LCD is working(shows bandFM and menu like Vol or Mute etc.)
Rotary encoder: ONLY the switch is working.
Serial COM works also with commands . 
Menu Button goes to A1. Since I dont have a seek Button i keep the code selection to A0.
Si4703 just produce noise. Even when stations are changed.
When i test it with Test4703 it works fine!
Can you please help me with this?? 
Thanks 

Andreas

     My Pinout goes like this:
 Arduino       | SI4703 sig   | LCD        |  KY040 Rotary Encoder
:----------:     | :-----------: | :--------: | :-----------:
 GND (black) | GND           | GND      |  GND
 3.3V (red)     | VCC            | -            |  -
 5V (red)       | -                  | VCC       |  VCC 5V
 A5 (yellow)  | SCLK           | SCLK      |  -
 A4 (blue)     | SDIO           | SDIO      |  - 
 D2               | RST             | -             |  - 
 A3 (white)   | -                  | -             |  DT  
 A2 (yellow) | -                  |  -            |  CLK
 A1 (green)  | -                  |  -            |  SW

Code is here!!! :

`///
/// \file LCDRadio.ino 
/// \brief Radio implementation including LCD output.
/// 
/// \author Matthias Hertel, http://www.mathertel.de
/// \copyright Copyright (c) 2014 by Matthias Hertel.\n
/// This work is licensed under a BSD style license.\n
/// See http://www.mathertel.de/License.aspx
///
/// \details
/// This is a full function radio implementation that uses a LCD display to show the current station information.\n
/// It can be used with various chips after adjusting the radio object definition.\n
/// Open the Serial console with 57600 baud to see current radio information and change various settings.
///
/// Wiring
/// ------
/// The necessary wiring of the various chips are described in the Testxxx example sketches.
/// The boards have to be connected by using the following connections:
/// 
/// Arduino port | SI4703 signal | RDA5807M signal
/// :----------: | :-----------: | :-------------:
/// GND (black)  | GND           | GND   
/// 3.3V (red)   | VCC           | VCC
/// 5V (red)     | -             | -
/// A5 (yellow)  | SCLK          | SCLK
/// A4 (blue)    | SDIO          | SDIO
/// D2           | RST           | -
///
/// More documentation and source code is available at http://www.mathertel.de/Arduino
///
/// History:
/// --------
/// * 05.08.2014 created.
/// * 06.10.2014 working.

#include <Wire.h>

#include <radio.h>
#include <RDA5807M.h>
#include <SI4703.h>
[... 457 lines cut ...]
  radio.checkRDS();

  // update the display from time to time
  if (now > nextFreqTime) {
    f = radio.getFrequency();
    if (f != lastf) {
      // don't display a Service Name while frequency is no stable.
      DisplayServiceName("        ");
      DisplayFrequency(f);
      lastf = f;
    } // if
    nextFreqTime = now + 400;
  } // if  

  if (now > nextRadioInfoTime) {
    RADIO_INFO info;
    radio.getRadioInfo(&info);
    lcd.setCursor(14, 0);
    lcd.print(info.rssi);
    nextRadioInfoTime = now + 1000;
  } // update

} // loop

// End.`

Beppi4U, 2020-12-05:
Ok! After some experimenting with this and that i found out : 

1.  I have to set up the "interrupt service routine" correctly. Means line 257 
     change this "ISR(PCINT2_vect"   to --->  that "ISR(PCINT1_vect"
     Now the rotary encoder works! But directions are reversed -- turn left and values 
     are increasing turning right values are decreasing. 
     Hope someone can tell me to change that!
2.  Uncommenting lines 290-291
    `PCICR  |= (1 << PCIE1);    // This enables Pin Change Interrupt 1 that covers the Analog input pins or Port C.
   PCMSK1 |= (1 << PCINT10) | (1 << PCINT11);  // This enables the interrupt for pin 2 and 3 of Port C`
     ...so rotary encoder setup seems to be complete!
      Comment out the lines for Mega2560 few lines down.

    BUT!!  Still no sound just noise!!!!  
               Setting Volume workes but on the LCD the displayed values are going nuts. 
   Would somebody please help me with that!?? Thanks a lot!
   Andreas

mathertel, 2023-02-07:
Please use the TESTxxx examples first to make sure the chip and setup works.

The interrupt code was changed in version 3.0.x

closed 2023-02-07 by mathertel