mathertel/Radio #50: Library not compiling on Arduino IDE (radio.cpp:426:16: error: 'class HardwareSerial' has no member named 'printf')

JuanLPerea, 2021-01-01:
Hi I'm testing your code on Arduino IDE 1.8.10 and there are some errors when compiling:
 radio.cpp:426:16: error: 'class HardwareSerial' has no member named 'printf'
 radio.cpp:443:16: error: 'class HardwareSerial' has no member named 'printf'
 radio.cpp:451:16: error: 'class HardwareSerial' has no member named 'printf'

I'm using your example code 'TestRDA5807M' with no modifications, and your library downloaded from github.

There are also many typedef declarations warnings, but the main problem is the code not compiling by the errors detailed.

Am I making some mistake or what could be the cause of this compiling errors?

Thank you in advance

Juan Luis

julianmcconnell, 2021-01-03:
What Arduino board are you compiling for?

JuanLPerea, 2021-01-05:
I'm trying to compile on Arduino Uno and Pro Mini 328P 5V with the same result. Also tried on different systems (Windows, Linux, Mac) and obtained identical error.

Anybody there had this error too?

Thanks and regards

julianmcconnell, 2021-01-05:
I don't see any references to printf in the example. All references are to print or println. Are you sure you didn't modify the example?

JuanLPerea, 2021-01-05:
Yes, the example is not modified. As you can see on the error message, the printf call is in 'radio.cpp' file, wich is a part of the 'radio' library. 
As I think know, with the '#include <radio.h>' the compiler builds the source code joining your code with the included library's code and generates machine language (binary) that can be executed by the Arduino board.
In this case, the error came from the library.
Thanks for your comment

taunoe, 2021-01-05:
When I look source code in file radio.cpp
Line 426: `Serial.printf("%02x ", d);`
Line 443: 'Serial.printf("[%d]", received);'
Line 451: `Serial.printf("%02x ", *d);`

JuanLPerea, 2021-01-05:
Exactly taunoe, in these lines in 'radio.cpp' are calling to a 'printf' function on the Serial.cpp base library of Arduino. And there is no printf function on Serial.cpp source code, then the compiler returns the error.
The question is, I saw a lot of people use mathertel radio library, and is used in many projects on the Web, but the didn't have this error? How?

taunoe, 2021-01-05:
I tested it on Arduino IDE and there are errors. Internet says that there is no printf function in Arduino IDE https://github.com/arduino/Arduino/pull/5938
When i build exact same code with PlatformIO then there is no errors.

JuanLPerea, 2021-01-05:
Thanks taunoe, this is a very smart solution, I will try it. 
Arduino IDE is very basic and until today is the only tool that I use to make my electronic projects, and I didn't know that can be programmed with PlattformIO.
Best regards!

mathertel, 2021-01-16:
in the HEAD of the master branch this is already fixed.

JuanLPerea, 2021-01-17:
Thanks Matthias, now the library compiles also on Arduino IDE. I have also installed Atom & PlatformIO as suggested taunoe and is a very nice IDE to program Arduino and now I'm programming with this tool I recommend it.
Best Regards

closed 2021-01-20 by JuanLPerea