mathertel/Radio #66: rds.attachServicenNameCallback(DisplayServiceName);

fungreenfox, 2023-05-18:
This line fails after updating the radio.h library.
rds.attachServicenNameCallback(DisplayServiceName);

exit status 1
'class RDSParser' has no member named 'attachServicenNameCallback'

Where is the documentation for those changes?

fungreenfox, 2023-05-18:
Found some in changelog..

closed 2023-05-18 by fungreenfox

z34pec, 2023-07-15:
Please comment more about this error. I am trying to find out why this error is present but as a newbie  I am not able to fix it. Thank you :)

fungreenfox, 2023-07-26:
Well.... I tried to update to newest Si4703 library, but did not have the RDS to work. It works with an older library so thats what is in use. To begin with, there was confusion about the uint8_t variables, that the logic tells me, they can easily be only assigned as int8_t. This caused a lot of trouble. Looking into the library, made me set those to uint8_t, and a lot of errors disappeared. As those variables do not seem to at any point be going below positive, it was not logic to set as unsigned. 
Might also be important to note, that if updating the librarys, the rdsparser should also follow same version.
The script is working atm with older library, and i am not sure, what exactly called that error. I just picked an issue up, since it has been a while.