mathertel/Radio #67: fail at bassboost

fungreenfox, 2023-05-18:
Only SI4703 is decleared in the script, however there are compiler warnings for SI4705:

Driveletter:\Arduino scripts\libraries\Radio\src\SI4705.cpp:229:34: error: variable or field 'setBassBoost' declared void
 void SI4705::setBassBoost(UNUSED bool switchOn) {
                                  ^
Driveletter:\Arduino scripts\libraries\Radio\src\SI4705.cpp:229:27: error: 'UNUSED' was not declared in this scope
 void SI4705::setBassBoost(UNUSED bool switchOn) {

mathertel, 2023-05-19:
What processor are you using ?

fungreenfox, 2023-05-21:
Esp32 Wroom 30 pins.
The problem has disappeared though.

fungreenfox, 2023-05-23:
However, when the volume is turned upwards / raised over 4 or 5 or so, the esp32 reboots. This happened only with the newer version of radio.h

fungreenfox, 2024-07-15:
Solved the issue.

closed 2024-07-15 by fungreenfox