mathertel/Radio #70: SI4703 hanging at Function setFrequency()

pr-boh, 2023-12-17:
Hello, I bought two breakout modules for Si4703 from Makershop.de. When testing with your
[image]
 library I have the following problem. In every example sketch, the software hangs when I call the setFrequency() function. I tried the examples with Arduino pro mini, with ESP8266 (D1 mini) and with ESP32. Same problem every time. When calling setFrequency(), the Sketch stops at _waitEnd(). I would be very grateful for a tip as to the cause of the problem.

Fishbone69, 2024-03-31:
Hi.  @mathertel @pr-boh   This seems to be a common issue with esp32

I am having the exact same issue using esp32-s3.  What's more, it isn't just this library.  I want to point you here:
https://forum.arduino.cc/t/esp32-si4703-fm-module-not-responding/1184277
and here:
https://github.com/pu2clr/SI470X/issues/4

I don't know if this is a compatibility issue with esp32-s3 or if there are just a bunch of us using esp32s that are making the same rookie mistake.  I'll provide some additional info.
for my setup, I am using
1.  External Reference Clock (from DS3231) - no crystal
2.  On Si4703 GPIO2 - not connected
3.  On Si4703 GPIO1 and GPIO3 = +3.3v (High)

I would be happy to troubleshoot.  Any help is appreciated.

cassou, 2024-06-14:
@pr-boh @Fishbone69 I had a similar issue, and in my case it was the power-supply (3V from my esp32 devboard) that was not powerfull enough to power the si4703 after the boot stage.

Adding an external 3.3 power supply for the si4703 fixed the issue.